me-no-dev/ESPAsyncTCP #10: Publish the library on PlatformIO

marvinroger, 2016-05-09:
Would you mind publishing this awesome library to PlatformIO? I am currently working on an async MQTT client, and it would be handy not having to download the dependency manually. Thanks!

me-no-dev, 2016-05-10:
I surely do not mind at all, but I am not a platformio user, so can you please point me in a direction where I can read on what needs to be done for the lib to be supported?

marvinroger, 2016-05-10:
Sure, just add a `library.json` file to the root of the repo:

````json
{
  "name": "ESPAsyncTCP",
  "keywords": "async, tcp, esp8266, esp32, client, server",
  "description": "Async TCP Library for ESP8266 Arduino",
  "authors":
  {
    "name": "<Your name>"
  },
  "repository":
  {
    "type": "git",
    "url": "https://github.com/me-no-dev/ESPAsyncTCP.git"
  },
  "frameworks": "arduino",
  "platforms": "espressif"
}
```

Then you just need to run `platformio lib register [HTTP_URL_TO_LIBRARY.JSON]`, and the crawler will automatically update the library on the registry whenever you push new commits, so once registered this is completely transparent.

I guess @ivankravets might even register the library for us. :smiley:

me-no-dev, 2016-05-10:
OK I created the json :) I guess I need platformio to ```platformio lib register [HTTP_URL_TO_LIBRARY.JSON]``` ? Could you do that since you have it installed and the jsons are already in the repositories for both ESPAsyncTCP and ESPAsyncWebServer?

marvinroger, 2016-05-10:
Well, to be honest I don't have it installed either, but I had a lot of requests asking me to put my own library on the registry. Ivan is the creator of PlatformIO, so he will probably run the appropriate commands to register both libs. ;)

ivankravets, 2016-05-10:
Hello from @PlatformIO!

Done! http://platformio.org/lib/show/305/ESPAsyncTCP

Happy Coding! 😊

marvinroger, 2016-05-10:
Awesome, thanks to both of you. :)

closed 2016-05-10 by marvinroger

me-no-dev, 2016-05-10:
@ivankravets any chance you add https://raw.githubusercontent.com/me-no-dev/ESPAsyncWebServer/master/library.json to the mix as well? Thank you

ivankravets, 2016-05-10:
@me-no-dev 

http://platformio.org/lib/show/306/ESPAsyncWebServer

Done! 👍 Thanks!