me-no-dev/ESPAsyncTCP #11: Compiling Error (even on fresh Arduino Sketch)

ccostel, 2016-05-25:
After I include the library in an empty Arduino sketch I get the following error: 

`AsyncPrinter.cpp:152:34: error: 'class cbuf' has no member named 'available'
   size_t available = _tx_buffer->available();`

Am I doing something wrong? Am I missing something?

me-no-dev, 2016-05-26:
You are using old version of the ESP8266-Arduino core. Update to the latest version and it will compile fine :)

ccostel, 2016-05-26:
Oh ok. I will give it a try tonight and let you know. Thanks for the quick reply.

ccostel, 2016-05-26:
Hey again! You were right I had to update my esp8266 library. I can compile again. Thank you!

I know it's a bit far fetched to ask but here goes. I am trying to use your library together with the https://github.com/Links2004/arduinoWebSockets library. But being new to this I can't understand how should I handle the code. Do you maybe have/know where I can find some example code?

me-no-dev, 2016-05-26:
use the async branch of the WebSocket library and change the define [here](https://github.com/Links2004/arduinoWebSockets/blob/async/src/WebSockets.h#L57) to enable async.
I imagine you are also running a web server on the ESP. If so, you might want to look into my AsyncWebServer library that also has AsyncWebSocket implementation (for full Async networking)

ccostel, 2016-05-26:
Hmm thanks. I was not aware the there was an async branch. I compared it on github with the master and they seem to be identical. I am sure I am missing something. I did enable the async in the Websocket.h but with my [code](https://github.com/ccostel/faraday-esp8266-websockets/blob/master/websocket.ino) (and your library included) I get a warning stating: 

`'void WebSocketsServer::loop()' is deprecated ... `

The warning happens right [here](https://github.com/Links2004/arduinoWebSockets/blob/async/src/WebSocketsServer.h#L54) . 

If a loop is not needed how would one handle the incoming websocket messages?

p.s. I am not running a web server. Just a wifi access point. I use javascript to send websocket packages.

I will also take a look at your AsyncWebServer. 

p.s. 2 I want to use your library to send lighting fast messages from a smartphone to control an electric skateboard :) and the sync websocket is actually chugging the esp8266.

marvinroger, 2016-05-26:
@ccostel this is the point of the asynchronous library, everything is handled in the "background" without the need for a `loop()` in the main loop. The whole interaction happens in the callbacks.

ccostel, 2016-05-26:
@marvinroger thanks for jumping in! I am definitely doing something wrong. I removed the loop() and made a call back function (sendMessage()) that sends back to the websocket client a string. 

Now I get this error: 
`ava.net.SocketException: Can't assign requested address
	at java.net.PlainDatagramSocketImpl.join(Native Method)
	at java.net.AbstractPlainDatagramSocketImpl.join(AbstractPlainDatagramSocketImpl.java:179)
	at java.net.MulticastSocket.joinGroup(MulticastSocket.java:323)
	at javax.jmdns.impl.JmDNSImpl.openMulticastSocket(JmDNSImpl.java:463)
	at javax.jmdns.impl.JmDNSImpl.<init>(JmDNSImpl.java:420)
	at javax.jmdns.JmDNS.create(JmDNS.java:81)
	at cc.arduino.packages.discoverers.NetworkDiscovery.inetAddressAdded(NetworkDiscovery.java:189)
	at cc.arduino.packages.discoverers.network.NetworkChecker.run(NetworkChecker.java:62)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
`

[Here's my code](http://pastebin.com/a3DhqQjY)

marvinroger, 2016-05-26:
Just restart your IDE, the error you're describing is not linked to your code. It's an Arduino bug that happens sometimes, when compiling.

ccostel, 2016-05-26:
I did restart the IDE. The error happens after the code is uploaded to the nodemcu. A colleagie suggestes that the nodemcu might be reseting itself (the on board led blinks once before the error appears in the ide). @marvinroger i think my code is the reason. Gonna try a few more things.

ccostel, 2016-05-26:
Guys thank you for your help! Now it all works. I am in the train now but i was able to find the problem. It was a yield() call in the websocket event listener. I will come back with a more elaborate solution to my issue. Thanks again, i am sendig websockets at 5ms speed :),

closed 2016-09-23 by me-no-dev