me-no-dev/ESPAsyncTCP #12: using Async connection for a Server - Clinets connection

mkeyno, 2016-06-07:
hi @me-no-dev  sorry to make the issue for my question , actually I've planned to set module as AP  and other module as Station mode and like IOT home , the server received and send command to those , Iggre told me I should use your lib instead of Esp8266WebServer lib , I really appreciated if you could guide me to what strategy should follow

me-no-dev, 2016-06-07:
you need web server or tcp server? Can you describe a bit better the use case?

mkeyno, 2016-06-07:
thanks @me-no-dev for reply soon , I want to build the Home IOT platform and use one ESP as AP server which can gather and command other wifi-sensor (5 to 10 modules) , I should make webpage on AP server so user can login and monitor(command) all activities, I estimate the volume of data is not more than one request per second , what do you suggest for such plan ? how about the Async MQTT?
I think this is not new idea and perhaps  many guys attempt to make home IOT with ESP modules, do you know any repo or lib?

closed 2016-07-25 by me-no-dev