me-no-dev/ESPAsyncTCP #13: SyncClient.ino  - The document has moved

RudyFiero, 2016-06-10:
I tried the SyncClient sketch and got this. I don't know what it should have done but I don't think this is what you got when you made the example. 

`HTTP/1.1 302 Found
Cache-Control: private
Content-Type: text/html; charset=UTF-8
Location: http://www.google.ca/?gfe_rd=cr&ei=dBNaV9XDJPKy8weD8qx4
Content-Length: 256
Date: Fri, 10 Jun 2016 01:10:12 GMT

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.ca/?gfe_rd=cr&amp;ei=dBNaV9XDJPKy8weD8qx4">here</A>.
</BODY></HTML>
{l`

me-no-dev, 2016-06-10:
yup, that is OK :) Google is forwarding you to http://www.google.ca/?gfe_rd=cr&ei=dBNaV9XDJPKy8weD8qx4

me-no-dev, 2016-06-10:
try the sketch to connect to www.google.ca instead

RudyFiero, 2016-06-10:
Make sense no. I'll give it a try.

closed 2016-07-25 by me-no-dev