me-no-dev/ESPAsyncTCP #14: library.properties still says 1.0.0

mozgy, 2016-07-06:
Please start versioning this library .. meaning move from 1.0.0 already ;)
Easier to update through Arduino IDE 'Managing Libraries'.

me-no-dev, 2016-07-06:
have been trying to implement the latest bigger changes in the web server and some required work on the TCP lib as well. I'm almost done implementing all I want to have as version 1 and will update and release properly. There were some really small changes that made big difference in some cases and would have been shame to update release for one line.
Need to test server and client destructions and so on, that might change more TCP code before release.

jeppo77, 2016-11-19:
hi
We're a couple of months further with more fixes, but Platformio's lib manager still downloads v1.0.0... Would a version update be suitable now?
thanks

me-no-dev, 2016-11-19:
yeah :) given some recent changes I think it's good time to actually bump the version. Maybe I'll find time to put some more examples and release, as that seems to be the only real nagging problem at the time.

closed 2019-10-05 by stale[bot]