me-no-dev/ESPAsyncTCP #15: AsyncClient example needed

IOTraul, 2016-07-23:
I'm trying to implement your AsyncClient non-blocking version of this library. Do you have an implementation example of this library ? i found the code but found no implementation example 

This is the Scenario/issue I have which I hope your library will help me with : 
Several sensors (atmega328 with NRF24L01) send data to one ESP8266 with NRF24L01 every around 2 secs
The sensors just send the data to the ESP8266, they don't wait for it to be available or wait for a confirmation from it, they just send the data.
The ESP8266 takes all the data received and places it in a queue.
Every 1 min approx the ESP8266 connects to a server and empties queue by posting all the data to the server.
While it posts the data, it won't listen the sensors so I will be missing some of the info.

michalfapso, 2016-08-15:
Hi @IOTraul, it seems that the SyncClient example provided in the repository, is actually easily adaptable to your scenario. You don't have to synchronously poll the client. Look at the example below. You can just send a client request to a server, then do any other processing (handling the nRF24L01) and when data arrive from the server in your client, you can read them anytime later.

```c++
#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#else
#include <ESP31BWiFi.h>
#endif
#include "ESPAsyncTCP.h"
#include "SyncClient.h"

const char* ssid = "**********";
const char* password = "**********";

SyncClient client;

void setup(){
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }
  if(!client.connect("fake-response.appspot.com", 80)){
    Serial.println("Connect Failed");
    return;
  }
  client.setTimeout(6);
}

void loop(){
  // Handle your nRF24L01 ...

  if (SOME_DATA_READY_FOR_SERVER) {
    // Send data to send to a remote server:
    if(client.printf("GET /api/?data={\"Hello\":\"World\"}&sleep=3 HTTP/1.1\r\nHost: fake-response.appspot.com\r\n\r\n") == 0){
      // TODO: handle the error
    }
  }

  // When you need to process the data returned from the server:
  if (client.connected() && client.available()) {
    Serial.println("gotData:");
    while(client.connected() && client.available()){
      Serial.write(client.read());
    }
  }
}
```

x-labz, 2016-08-18:
same problem here: #18

closed 2019-10-05 by stale[bot]