me-no-dev/ESPAsyncTCP #17: ESPAsyncTCP connect to dns hostname from async callback fails

marvinroger, 2016-08-17:
Moving the issue from https://github.com/marvinroger/async-mqtt-client/issues/18, as requested :)

me-no-dev, 2016-08-19:
done :) https://github.com/me-no-dev/ESPAsyncTCP/commit/cc4d1ac2b76510465dba10392202a73a2dbc772c
give it a go ;)
BTW i'm looking into running axTLS async (no success yet)

marvinroger, 2016-08-21:
Fixed, thanks!
Oh you'll succeed, I am not worried. axTLS 2.0 brings TLS 1.2 moreover, so perfect timing. :)

closed 2016-08-21 by marvinroger

me-no-dev, 2016-08-22:
@marvinroger check this out: https://github.com/me-no-dev/ESPAsyncTCP/commit/110a997a32a78f42a648095c6340ca201c2e5333

client->connect(host, port, true) for SSL

me-no-dev, 2016-08-22:
requires latest git of ESP8266 arduino

me-no-dev, 2016-08-22:
```cpp

static AsyncClient * aClient = NULL;

const char *sslHost = "192.168.254.16";
const uint16_t sslPort = 443;

static void securePrintInfo(SSL *ssl){
  if(!ssl){
    return;
  }
  const char *common_name = ssl_get_cert_dn(ssl, SSL_X509_CERT_COMMON_NAME);
  if (common_name) {
    Serial.printf("COMMON NAME: %s\r\n", common_name);
  }

  Serial.printf("CIPHER: ");
  switch (ssl_get_cipher_id(ssl)){
    case SSL_AES128_SHA:
      Serial.printf("AES128-SHA");
      break;

    case SSL_AES256_SHA:
      Serial.printf("AES256-SHA");
      break;

    case SSL_RC4_128_SHA:
      Serial.printf("RC4-SHA");
      break;

    case SSL_RC4_128_MD5:
      Serial.printf("RC4-MD5");
      break;

    default:
      Serial.printf("Unknown - %d", ssl_get_cipher_id(ssl));
      break;
  }

  Serial.printf("\r\n");

  int i;
  const uint8_t *session_id = ssl_get_session_id(ssl);
  int sess_id_size = ssl_get_session_id_size(ssl);
  if (sess_id_size > 0){
    Serial.printf("SESSION: ");
    for (i = 0; i < sess_id_size; i++){
      Serial.printf("%02x", session_id[i]);
    }
    Serial.printf("\r\n");
  }
}

void runAsyncClient(){
  if(aClient)//client already exists
    return;

  aClient = new AsyncClient();
  if(!aClient)//could not allocate client
    return;

  aClient->onError([](void * arg, AsyncClient * client, int error){
    Serial.println("Connect Error");
    aClient = NULL;
    delete client;
  }, NULL);

  aClient->onConnect([](void * arg, AsyncClient * client){
    Serial.println("Connected");
    securePrintInfo(client->getSSL());
    aClient->onError(NULL, NULL);

    client->onDisconnect([](void * arg, AsyncClient * c){
      Serial.println("Disconnected");
      aClient = NULL;
      delete c;
    }, NULL);

    client->onData([](void * arg, AsyncClient * c, void * data, size_t len){
      Serial.print("\r\nData: ");
      Serial.println(len);
      uint8_t * d = (uint8_t*)data;
      for(size_t i=0; i<len;i++)
        Serial.write(d[i]);
    }, NULL);

    //send the request
    char m[256];
    sprintf(m, "GET /test.htm HTTP/1.0\r\nHost: %s\r\n\r\n", sslHost);
    int wrote = client->write(m, strlen(m));
    Serial.printf("Sent: %u => %d\r\n", strlen(m), wrote);
  }, NULL);

  if(!aClient->connect(sslHost, sslPort, true)){
    Serial.println("Connect Fail");
    AsyncClient * client = aClient;
    aClient = NULL;
    delete client;
  }
}

```

marvinroger, 2016-08-24:
Awesome! I will play with this as soon as I am back from vacation. Well done :)