me-no-dev/ESPAsyncTCP #19: Compile Errors after update of ESPAsyncTCP

idolpx, 2016-08-25:
Everything was working great then the library was updated in PlatformIO and I'm now getting the following errors.

Please advise.

```
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o):(.bss.default_private_key+0x0): multiple definition of `default_private_key'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):(.bss.default_private_key+0x0): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o):(.bss.default_private_key_len+0x0): multiple definition of `default_private_key_len'

.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):(.bss.default_private_key_len+0x0): first defined here
/Users/jjohnston/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: Warning: size of symbol `default_private_key_len' change
d from 2 in .pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o) to 4 in .pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o)
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o):(.bss.default_certificate+0x0): multiple definition of `default_certificate'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):(.bss.default_certificate+0x0): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o):(.bss.default_certificate_len+0x0): multiple definition of `default_certificate_len'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):(.bss.default_certificate_len+0x0): first defined here
/Users/jjohnston/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: Warning: size of symbol `default_certificate_len' change
d from 2 in .pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o) to 4 in .pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o)
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_port_read':
WiFiClientSecure.cpp:(.text.ax_port_read+0x14): multiple definition of `ax_port_read'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):tcp_axtls.c:(.text.ax_port_read+0x14): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_port_write':
WiFiClientSecure.cpp:(.text.ax_port_write+0x1c): multiple definition of `ax_port_write'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):tcp_axtls.c:(.text.ax_port_write+0xc): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_get_file':
WiFiClientSecure.cpp:(.text.ax_get_file+0x0): multiple definition of `ax_get_file'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):tcp_axtls.c:(.text.ax_get_file+0x8): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_wdt_feed':
WiFiClientSecure.cpp:(.text.ax_wdt_feed+0x8): multiple definition of `ax_wdt_feed'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(tcp_axtls.o):tcp_axtls.c:(.text.ax_wdt_feed+0x0): first defined here
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/d1_mini/firmware.elf] Error 1

```

andriyadi, 2016-08-26:
I'm experiencing it too.

me-no-dev, 2016-08-26:
you need the git version of ESP8266 Arduino, but if you pull the latest lib it should work too as those things are now commented (SSL related errors)

closed 2016-09-23 by me-no-dev