me-no-dev/ESPAsyncTCP #21: Need example of sending TCP packets to a listener

Erik-k, 2016-09-14:
Hello, I want to send data (probably in the form of TCP packets) to another device on a WAN which is listening with $netcat -l <port>. I can't understand the SyncClient example well enough to figure this out. 

Sorry for begging but will you please give an example of how to craft a TCP packet and send it to another IP:port address?

me-no-dev, 2016-09-23:
Will you connect to that host every time you want to send a packet or will you stay connected to it?

NeilSpp, 2016-10-06:
void loop()
{
	if(millis() - previousTime > 200)
	{
		previousTime = millis();

		if(aClient1)//client already exists
	    	return;

		aClient1 = new AsyncClient();

		if(!aClient1)//could not allocate client
			return;

		aClient1->onConnect([](void * arg, AsyncClient * client) {

			client->write("tcp test string.");

			aClient1 = NULL;
			delete client;

		}, NULL);

		if(!aClient1->connect("192.168.0.100", 80))
		{
			AsyncClient * client = aClient1;
			aClient1 = NULL;
			delete client;
		}

		Serial.print("system_get_free_heap_size ");
                Serial.println(system_get_free_heap_size());
	}
}
  
  
This code can work, but after a period of time, system heap memory will continue to decline to 0.
How to correct close connect and free system heap memory?

Erik-k, 2016-11-17:
I'll stay connected to it.

closed 2019-10-05 by stale[bot]