me-no-dev/ESPAsyncTCP #27: Cannot reconnect after disconnecting

yaqwsx, 2016-10-09:
I have a following piece of code (shortened, some functions are missing):
```CPP
SyncClient socket;

void connect_to_ap(const char* ssid, const char* password) {
  if (WiFi.status() == WL_CONNECTED)
    return;
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
  }
}

bool connect_to_server() {
  while(socket.connect(server, port)) {
    if (WiFi.status() != WL_CONNECTED)
      return false;
    delay(50);
  }

  return true;
}

void setup() {
  while(true) { 
    Serial.begin(115200);
    init_gpio();
    connect_to_ap(NET_SSID, NET_PASSWORD);
    setup_ota();
    Serial.print("Connecting to server... ");
    if (!connect_to_server()) {
      Serial.println("failed");
      continue;
    }
    Serial.println("succeeded");
    break;
  } 
}

void loop() {
  ArduinoOTA.handle();
  yield();
  if (!socket.connected())
    setup();

  // Do some work
}
```

Basically, it should connect to WiFi, connect to server and restore the connection, when lost. The first connection is Ok, however when the server closes the connection and `socket.connect(server, port)` is called for the second time, it never returns (even the server starts accepting new connections). Calling `socket.stop()` does not help, creating new SyncClient via `socket = SyncClient()` causes crash.

I think this is not the expected behavior and it is a probably a bug in the library (if I am not doing anything wrong).

me-no-dev, 2016-10-10:
I see while, while, while... calling setup in loop
```while(socket.connect(server, port))``` will connect to new socket every 50 ms... so on
many, many things wrong in your code ;)

yaqwsx, 2016-10-14:
I am sorry, for a typo in the shortened version - of course there is `while(!socket.connect(server, port))` in my code (I fixed it in the original post).

What exactly is wrong in the code? I can't see it. `setup` initializes Serial, GPIO, connects to AP and then tries to connect to server - if something goes wrong (e.g. network is lost during connection to server), setup procedure is repeated. After everything is established correctly, `loop` continues. If anything goes wrong in the loop, connection is restored in the same manner.

me-no-dev, 2016-11-13:
yeah looking at the code now again it should have generally worked. I am not used to being asked for SyncClient :) so I assumed Async which makes your code invalid
I have an idea though.
You can try changing !client->disconnecting() with client->connecting() [here](https://github.com/me-no-dev/ESPAsyncTCP/blob/master/src/SyncClient.cpp#L70) and [here](https://github.com/me-no-dev/ESPAsyncTCP/blob/master/src/SyncClient.cpp#L91)
Maybe last connection ends up in some weird state

closed 2019-10-05 by stale[bot]