me-no-dev/ESPAsyncTCP #29: AsyncClient: _poll before _rcv causes false timeout

Nubosch, 2016-10-12:
_poll can be called from stack before _rcv,
cause _rx_last_packet is initialized to 0 in constructor
_ poll will trigger the the timeout instantly.

this happens frequently in my setup since WebServer sets timeout.
https://github.com/me-no-dev/ESPAsyncWebServer/commit/d935b1d740a1e15c800ce16a2d458e660e32382b

initializing _rx_last_packet to millis() in constructor solves the problem for me,
but I'm not sure if this is the best solution for the issue.

me-no-dev, 2016-11-13:
aah I have fixed it and not reported :) https://github.com/me-no-dev/ESPAsyncTCP/commit/e074eaac39ca6318ac96801f48618bb59864ccc7

closed 2016-11-13 by me-no-dev

Nubosch, 2016-11-15:
finally managed to switch my platformio lib to upstream git.
thanks for the fix.