me-no-dev/ESPAsyncTCP #36: Adapting my code from regular WifiClient to AsyncClient

miguelpom, 2017-01-11:
Hello, everyone!

I'm working on a project that needs a non-blocking connect() function to use in a ESP8266 working as a wifi client. 

Every program cycle, my ESP reads some analog data from a matrix of sensors. Whenever the matrix remains "calm" (with no changes on the sensors) for at least 2 seconds, the ESP will connect to the wifi server and send its accumulated data.
The problem is that the WifiClient.connect() function takes up to 5 seconds trying to reach the server. During the connection process, I can lose critical reading from my sensors matrix. I need this time to be shorter.

Following the advise of @me-no-dev, I am adapting my code to the ESPAsyncTCP library. However, the present examples in the Issues section do not explain the "use" of the "non-blocking feature" of AsyncClient and/or AsyncPrinter clearly (or so I think). 
I am a newcomer to these type of projects and the lack of documentation in the ESP8266 github libraries doesn't help much either. 
For example, what's the equivalent function for WifiClient.available() in the AsyncClient class? How do you poll the client's buffer?

Anyway, I am looking for some example or documentation that explains the differences between the "Sync" and the "Async" modes of WiFi communication from the client's point of view. 

Any help will be appreciated. Thank you so much!

me-no-dev, 2017-01-11:
Async is nothing like Sync :)

You do not have available(), instead you have onData callback to handle incoming data as it comes.
Basically the whole thing is event based and you have callback for each different event.
For example:
- You start with connect(host, port, onConnectCallback) and attaching onError callback if connect fails.
- Once you have connected, you can attach the rest of the callbacks and start sending and receiving data.
- Once you get onDisconnect or you close the client, you will no longer get onData callbacks and you can not send data.
- That's about it.

me-no-dev, 2017-01-11:
Reading about TCP connection flow could help you immensely in understanding what each callback is and when it's used/called. You can not send as much data as you want, because you are sending packets and separating a large chunk is now your job to do. After you send a packet, you need to wait for onAck callback to signal that the other end got the packet. And there is surely more... Async is not easy in any way.

ompurwar, 2018-01-21:
@me-no-dev Hi, 
Can I get some reference how to use this library to create an HTTP connection to a server in the most reliable way? please provide some links.
Thanks in advance.

closed 2019-10-05 by stale[bot]