me-no-dev/ESPAsyncTCP #37: Is ESP32 supported?

n0f8r, 2017-01-18:
Links2004 over at the arduinoWebSockets library thought that ESPAsyncTCP works on ESP32, but I've just had a pile of compile errors trying to include ESPAsyncTCP.

I've not been working with Arduino long and C is not my first language, so perhaps there's something obvious I need to do? The compile errors look like a conflict between different versions of a network library. For instance (and loads more like that) : 

C:\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:113:36: error: macro "ip_route" requires 2 arguments, but only 1 given
   netif* interface = ip_route(&addr);

williamesp2015, 2017-02-28:
PlatformIo announced that ESPAsyncTCP is compatible with ESP32 but I got errors:

src\ESPAsyncTCP.cpp:468:6: error: prototype for 'void AsyncClient::_s_dns_found(const char*, ip_addr_t*, void*)' does not match any in class 
'AsyncClient'
void AsyncClient::_s_dns_found(const char *name, ip_addr_t *ipaddr, void *arg){
^
In file included from src\ESPAsyncTCP.cpp:24:0:
src\ESPAsyncTCP.h:100:17: error: candidate is: static void AsyncClient::_s_dns_found(const char*, ip_addr*, void*)
static void _s_dns_found(const char *name, struct ip_addr *ipaddr, void *arg);

^
src\ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getRemoteAddress()':
src\ESPAsyncTCP.cpp:566:26: error: 'ip_addr_t {aka struct _ip_addr}' has no member named 'addr'
return _pcb->remote_ip.addr;
^
src\ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':
src\ESPAsyncTCP.cpp:578:25: error: 'ip_addr_t {aka struct _ip_addr}' has no member named 'addr'
return _pcb->local_ip.addr;
^
src\ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
src\ESPAsyncTCP.cpp:824:14: error: 'ip_addr_t {aka struct _ip_addr}' has no member named 'addr'
local_addr.addr = (uint32_t) _addr;
^
*** [.pioenvs\esp32dev\src\ESPAsyncTCP.o] Error 1

me-no-dev, 2017-02-28:
@ivankravets is that true? I never said it's compatible with ESP32?

n0f8r, 2017-02-28:
@me-no-dev I'm not sure where the idea of ESP32 compatibility originated - but the dev behind the arduino web sockets library that I was using on ESP8266 also thought it was ESP32 compatible - of course I also ran into issues ...

Links2004 => "if I remember correct ESPAsyncTCP supports ESP32."

See here https://github.com/Links2004/arduinoWebSockets/issues/161

me-no-dev, 2017-02-28:
it was true for ESP31B, but for ESP32 things are much much different.

williamesp2015, 2017-02-28:
I have Websocket with ESPAsyncTCP and I need to use ESP32. Any help would be greatly appreciated.

me-no-dev, 2017-02-28:
I have ESP32 and LOOOTS of work before I get to Async (which is my own project rather than my food paying job)

ivankravets, 2017-02-28:
@me-no-dev That is something new 😊 

> PlatformIo announced that ESPAsyncTCP is compatible with ESP32 but I got errors

@williamesp2015 please sorry, who did announce? Please be careful with "Compatibility" information on a library page. See 

* http://platformio.org/lib/show/305/ESPAsyncTCP

n0f8r, 2017-02-28:
@ivankravets  it's right there in the title tagline on the page you linked "Asynchronous TCP Library for ESP8266 and **ESP32**"

ivankravets, 2017-02-28:
@me-no-dev gotcha! 😆 

* https://github.com/me-no-dev/ESPAsyncTCP/blame/master/library.json

me-no-dev, 2017-03-01:
damned I thought I cleaned all of those...

me-no-dev, 2017-03-01:
fixed!

timpur, 2017-07-26:
Sorry to ask this question, But do you think this lib some day soon will be supported to work on the esp32?
Tim.

camilorodegheri, 2017-08-06:
Hi, got a first version working on ESP32. There is more work to be done, but sharing with people to be evaluated.
#48

me-no-dev, 2017-09-08:
Look [HERE](https://github.com/me-no-dev/AsyncTCP)

closed 2017-09-08 by me-no-dev