me-no-dev/ESPAsyncTCP #41: SyncClient::connect hanging without end

mjoey89, 2017-04-21:
Hi,

I think I found a bug in SyncClient. I tried to use SyncClient::connect(const char *host, uint16_t port) function, but when the host was unreachable (or it's port not opened), the function did not return at all. Kept hanging.

I think the problem might be with this while loop inside the function:
`while(_client != NULL && !_client->connected() && !_client->disconnecting())
      delay(1);`

I figured out, that the _state variable of AsyncClient under the hood remains either 0 or 10, so it looks like it does not change from disconnected to connecting (or to anything else) state at all. Nevertheless I can see in Wireshark, that ESP is sending approx. 5-6 times the SYN message (the port on my computer is was not opened, no SYN-ACK response).

When the host and port is reachable the connect() works without any problem.

Best Regards

me-no-dev, 2017-04-21:
I guess that loop just needs a timeout :)

mjoey89, 2017-04-21:
Maybe :)
But I do not get it, why _state var does not change to at least 1 when the connecting is obviously "in progress", hence the SYN was already sent. So maybe the bug is actually at the state machine handling :)

closed 2019-10-05 by stale[bot]