me-no-dev/ESPAsyncTCP #44: send on WiFiClient in AsyncClient OnData problem!

mcr-ksh, 2017-05-17:
Hello,

thanks for this library. Seems a step towards the right direction as the performance is really bad without it. However, I have trouble getting to run the function in a right way.

I have a strange behavior when sending data through WiFiClient.
I receive data from a callback function onData() and send the data through a WiFiclient socket.

`
    client->onData([](void *arg, AsyncClient *c, void *data, size_t len) {
        Serial.print("async_recv(): onData len = %d\n", len);
        if ( send_sock(&client, (const uint8_t *)data, len) != len )
        {
           Serial.print("async_recv(): onData send failed\n");
        }
    }, NULL);
`

The client->write function returns before hand, see blow:

```
send_sock(): -->
:wr
:ww
send_sock(): FAILED: send 0 of amount of 420 bytes. connected 1. available 0
_send_sock(): 0 <--_
**:sent 420**
```

thus, the function returns 0 written bytes as a return code, and the kernel does acknowledge the ":send 420"
at a later point. Here an example of a successful send in sync:

```
send_sock(): -->
:wr
**:sent 10**
:ww
_send_sock(): 10 <--_
```

What could cause this behavior?

Best
M

me-no-dev, 2017-05-17:
what is send_sock() ?

me-no-dev, 2017-05-17:
oh... you want to send data to a normal client from Async callback? tat is not possible ;) regular client uses delay and yield and those are not permitted in Async

mcr-ksh, 2017-05-17:
well, that explains it. I wanted to use two async sockets if possible but doing protocol work in two async sockets is nuts.

me-no-dev, 2017-05-17:
Better do all synch work in the loop and set flags in the async callbacks

closed 2019-10-05 by stale[bot]