me-no-dev/ESPAsyncTCP #45: SyncClient example with SSL fingerprint check

tomaszbabiuk, 2017-06-02:
Hello

I really like your projects. I was able to start SSL using AsyncWebServer and I'm really impressed - finally my local connections are secured as they should.  What I'm really missing is AsyncClient with SSL example. Checking certificate fingerprint (like WiFiClientSecure from esp8266/arduino) whould be more than enough.

tomaszbabiuk, 2017-06-03:
Ok, I've figured this out.

I've used async client instead because it has getSSL() method that returns ssl context.
Having that I'm able to use methods from axTLS() for validation like: ssl_get_cert_dn or ssl_match_fingerprint.

There're also examples in WiFiClientSecure.cpp class on how to validate certificate (from adruino/esp8266).

I hope my comment will help someone to create safe and async http client :)

closed 2017-06-03 by tomaszbabiuk

pjunni, 2018-02-23:
@tomaszbabiuk thanks for your tips. I'm still not entirely sure how to switch from using WiFiClientSecure to using fingerprint with ESPAsyncTCP. Any chance to include a sample here https://github.com/me-no-dev/ESPAsyncTCP/tree/master/examples/?

objarni, 2019-01-28:
I concur with @pjunni @tomaszbabiuk , would be great if you could "translate" the WiFiClientSecure example [1] from ArduinoSDK to ESPAsyncTCP lib! How much money do you want, I will give you $$ :)

[1] https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino