me-no-dev/ESPAsyncTCP #50: Unable to contact server over http after enabling SSL for MQTT

gjt211, 2017-08-28:
I recently changed my code to use the Async libraries and am exceptionally happy with the results. My sensors are much more stable and don't give me much trouble at all.

I have a problem with a http request that is used for OTA updates.

My web server has two different certificates. One for the web site (HTTPS) which is renewed every year, and one for MQTTs which is renewed less frequently. (This means different fingerprints)

My sensors connect to the MQTT broker using SSL, and this is working fine.
The problem I have is that the location on the server for the OTA updates is setup for HTTP only, not HTTPS, and even if it was HTTPS the certificate fingerprint would not match because MQTT is using a different certificate.

Using non Async libraries this has been working well for many months, but now that I changed to using the Async libraries I get the result `Error (-1): HTTP error: connection refused` when the sensor performs a firmware update check. I can only assume that it is trying to connect using HTTPS, and not plain HTTP.

The address passed to the httpUpdate call is `http://servername.com/firmware/sensormodel/checkfirmware.php`

I guess what I am asking, is how to send a HTTP request even though I have SSL enabled?

Thanks in advance.

gjt211, 2017-08-28:
I should add, if I call my `httpUpdate();` function before the following in my `setup()` function the update works every time, but obviously only when the device is powered up, resets or reboots.

So simply, httpUpdate works before this code, but fails after it with a `connection refused` error.
```
  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(configuration.shost, hport);
  #ifdef ASYNC_TCP_SSL_ENABLED
    mqttClient.setSecure(MQTT_SECURE);
    if (MQTT_SECURE) {
      mqttClient.addServerFingerprint((const uint8_t[])MQTT_SERVER_FINGERPRINT);
    }
  #endif
```

jeroenst, 2018-11-21:
Any news on this? I can connect async to an unencrypted mqtt server but not via ssl. It connects, and waits for data I think and after a while the server disconnects because no handshake is done by the client.

gjt211, 2018-11-23:
@jeroenst I think your issue is different to mine.
I can connect async using ssl fine.
My problem is that once I connect to MQTT using ssl, I can't then send a non ssl http request.

gjt211, 2018-12-18:
Closing doe to no responses.

closed 2018-12-18 by gjt211