me-no-dev/ESPAsyncTCP #60: Now can't build with Core 2.3.0?

JamesNewton, 2017-10-18:
Just started trying to use this library yesterday, and I'm thinking your fixes to support 2.4.0 might have broken support for 2.3.0?

I really don't want to upgrade to 2.4.0 before it's had a chance to bleed a bit...

...I guess I will try to download the prior version of your library, but figured I should mention this first. 

````
D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

                            ^

In file included from D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]

 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;

                  ^

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

     tcp_err(_pcb, &_s_error);

                            ^

In file included from D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]

 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;

                  ^

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]

     tcp_poll(_pcb, &_s_poll, 1);

                               ^

In file included from D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

   tcp_err(pcb, &_s_error);
[... 114 lines cut ...]

D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]

   tcp_accept(_pcb, &_s_accept);

                              ^

In file included from D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.1 in folder: C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient 
Using library ESPAsyncTCP-master at version 1.1.0 in folder: D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncTCP-master 
Using library ESPAsyncWebServer-master at version 1.1.0 in folder: D:\Users\MyUser\Documents\Arduino\libraries\ESPAsyncWebServer-master 
Using library Ticker at version 1.0 in folder: C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker 
Using library EEPROM at version 1.0 in folder: C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\EEPROM 
Using library Hash at version 1.0 in folder: C:\Users\MyUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Hash 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
````

JamesNewton, 2017-10-18:
Yeah, commit 9b0cc37cd7bdf4b7d17a363141d2e988aa46652c builds under 2.3.0

People who don't sit on the bleeding edge of everything might want to know.

Uksa007, 2017-10-18:
I have similar errors under 2.3.0 I have changed the library.json back to commit 9b0cc37 but still have the errors, is there something else I need to do?

xoseperez, 2017-10-20:
It would have been better to commit 036ea44 to a "staging" branch and keep master compatible with stable 2.3.0. I'm too receiving issues in ESPurna firmware and fauxmoESP library due to this...

me-no-dev, 2017-10-20:
it would have actually been best to have some sort of define switch that could be used to make it compatible with both but seems that the lwip headers were not updated together with the lib :(

kherNI, 2017-10-20:
I'm running 2.4.0 and I'm getting the same errors.

me-no-dev, 2017-10-20:
@kherNI how come? Changes were checked. Maybe you are running ssl?

kherNI, 2017-10-20:
I just tried with the standard example.  The Board Manager has : https://github.com/esp8266/Arduino/releases/download/2.4.0-rc2/package_esp8266com_index.json

Here is the compiler message:
`C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:89:6: warning: unused parameter 'c' [-Wunused-parameter]

 void AsyncPrinter::_onConnect(AsyncClient *c){

      ^

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:183:47: warning: unused parameter 'c' [-Wunused-parameter]

   _client->onPoll([](void *obj, AsyncClient* c){ ((AsyncPrinter*)(obj))->_sendBuffer(); }, this);

                                               ^

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:183:47: warning: unused parameter 'c' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:184:73: warning: unused parameter 'c' [-Wunused-parameter]

   _client->onAck([](void *obj, AsyncClient* c, size_t len, uint32_t time){ ((AsyncPrinter*)(obj))->_sendBuffer(); }, this);

                                                                         ^

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:184:73: warning: unused parameter 'len' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:184:73: warning: unused parameter 'time' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:184:73: warning: unused parameter 'c' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:184:73: warning: unused parameter 'len' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:184:73: warning: unused parameter 'time' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:186:71: warning: unused parameter 'c' [-Wunused-parameter]

   _client->onData([](void *obj, AsyncClient* c, void *data, size_t len){ ((AsyncPrinter*)(obj))->_onData(data, len); }, this);

                                                                       ^

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:186:71: warning: unused parameter 'c' [-Wunused-parameter]

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:73:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:23:0:

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:74:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

                            ^

In file included from C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:23:0:

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]

 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;

                  ^
[... 181 lines cut ...]
C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:835:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]

   tcp_accept(_pcb, &_s_accept);

                              ^

In file included from C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:23:0:

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: At global scope:

C:\Users\User\Documents\Arduino\hardware\espressif\esp8266\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:895:6: warning: unused parameter 'err' [-Wunused-parameter]

 long AsyncServer::_accept(tcp_pcb* pcb, long err){

      ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
`

me-no-dev, 2017-10-20:
Something is wrong. This can not be 2.4.0

kherNI, 2017-10-20:
You're right.  I just did a git clone for the standard ESP8266 and this repository and now it compiles fine.  Sorry about that.  Thanks for all your help on this platform!

universam1, 2017-11-01:
could you @me-no-dev maybe add a condition based on `const char * getSdkVersion()` to solve this here. We got so many devs having issue because of that.

andig, 2017-11-02:
Fixed in master and can be closed?

fdarling, 2017-11-12:
I still get the compilation errors when using the latest master branch of this project and the following setup:

Ubuntu 16.04 (amd64)
Arduino IDE 1.8.5 (also tested 1.8.2)
esp8266 v2.3.0 installed via the boards manager

I wouldn't consider this closed, it's a problem because it doesn't work "out of the box" with a standard installation!

Is this a matter of putting some macros somewhere to select between different function signatures?

fdarling, 2017-11-13:
The root problem seems to be the hard-coding of "long" instead of "err_t". err_t is defined in lwip/err.h, and in my case maps to "s8" (signed char).

I created a patch, but I'm a noob to git/github, so I don't know the proper way to upload it. I put it inline here wrapped in a triple backtick block. If this work for you, please speak up and let's get this into the master branch!

```diff --git a/src/ESPAsyncTCP.cpp b/src/ESPAsyncTCP.cpp
index 593d6d6..433d318 100644
--- a/src/ESPAsyncTCP.cpp
+++ b/src/ESPAsyncTCP.cpp
@@ -478,23 +478,23 @@ void AsyncClient::_s_dns_found(const char *name, const ip_addr *ipaddr, void *ar
   reinterpret_cast<AsyncClient*>(arg)->_dns_found(ipaddr);
 }
 
-long AsyncClient::_s_poll(void *arg, struct tcp_pcb *tpcb) {
+err_t AsyncClient::_s_poll(void *arg, struct tcp_pcb *tpcb) {
   return reinterpret_cast<AsyncClient*>(arg)->_poll(tpcb);
 }
 
-long AsyncClient::_s_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *pb, long err) {
+err_t AsyncClient::_s_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *pb, err_t err) {
   return reinterpret_cast<AsyncClient*>(arg)->_recv(tpcb, pb, err);
 }
 
-void AsyncClient::_s_error(void *arg, long err) {
+void AsyncClient::_s_error(void *arg, err_t err) {
   reinterpret_cast<AsyncClient*>(arg)->_error(err);
 }
 
-long AsyncClient::_s_sent(void *arg, struct tcp_pcb *tpcb, uint16_t len) {
+err_t AsyncClient::_s_sent(void *arg, struct tcp_pcb *tpcb, uint16_t len) {
   return reinterpret_cast<AsyncClient*>(arg)->_sent(tpcb, len);
 }
 
-long AsyncClient::_s_connected(void* arg, void* tpcb, long err){
+err_t AsyncClient::_s_connected(void* arg, void* tpcb, err_t err){
     return reinterpret_cast<AsyncClient*>(arg)->_connected(tpcb, err);
 }
 
@@ -973,7 +973,7 @@ long AsyncServer::_accept(tcp_pcb* pcb, long err){
   return ERR_OK;
 }
 
-  long AsyncServer::_s_accept(void *arg, tcp_pcb* pcb, long err){
+  err_t AsyncServer::_s_accept(void *arg, tcp_pcb* pcb, err_t err){
     return reinterpret_cast<AsyncServer*>(arg)->_accept(pcb, err);
   }
 
diff --git a/src/ESPAsyncTCP.h b/src/ESPAsyncTCP.h
index 1e72dfd..2954b37 100644
--- a/src/ESPAsyncTCP.h
+++ b/src/ESPAsyncTCP.h
@@ -26,6 +26,7 @@
 #include "IPAddress.h"
 #include <functional>
 #include "lwip/init.h"
+#include "lwip/err.h"
 
 class AsyncClient;
 
@@ -97,11 +98,11 @@ class AsyncClient {
 #else
     void _dns_found(const ip_addr *ipaddr);
 #endif
-    static long _s_poll(void *arg, struct tcp_pcb *tpcb);
-    static long _s_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *pb, long err);
-    static void _s_error(void *arg, long err);
-    static long _s_sent(void *arg, struct tcp_pcb *tpcb, uint16_t len);
-    static long _s_connected(void* arg, void* tpcb, long err);
+    static err_t _s_poll(void *arg, struct tcp_pcb *tpcb);
+    static err_t _s_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *pb, err_t err);
+    static void _s_error(void *arg, err_t err);
+    static err_t _s_sent(void *arg, struct tcp_pcb *tpcb, uint16_t len);
+    static err_t _s_connected(void* arg, void* tpcb, err_t err);
 #if LWIP_VERSION_MAJOR == 1
     static void _s_dns_found(const char *name, struct ip_addr *ipaddr, void *arg);
 #else
@@ -234,14 +235,14 @@ class AsyncServer {
 
   protected:
     long _accept(tcp_pcb* newpcb, long err);
-    static long _s_accept(void *arg, tcp_pcb* newpcb, long err);
+    static err_t _s_accept(void *arg, tcp_pcb* newpcb, err_t err);
 #if ASYNC_TCP_SSL_ENABLED
     int _cert(const char *filename, uint8_t **buf);
     long _poll(tcp_pcb* pcb);
     long _recv(tcp_pcb *pcb, struct pbuf *pb, long err);
     static int _s_cert(void *arg, const char *filename, uint8_t **buf);
-    static long _s_poll(void *arg, struct tcp_pcb *tpcb);
-    static long _s_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *pb, long err);
+    static err_t _s_poll(void *arg, struct tcp_pcb *tpcb);
+    static err_t _s_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *pb, err_t err);
 #endif
 };
 
```

whitebird, 2017-11-16:
@fdarling that worked perfectly for me, thanks!

max5555, 2017-11-19:
@whitebird can you upload the patched variant somewhere?

andig, 2017-11-19:
@me-no-dev would it make sense to change `long` to `err_t`?

fdarling, 2017-11-19:
I think we should be using `err_t` in favor of `long` or `s8`. The idea is for consistency/portability/version compatibility, our callback function signatures should match the libraries signature typedef exactly, and not have the underlying/equivalent data-types substituted.

`ESPAsyncTCP.cpp` includes the `"lwip/tcp.h"` lwIP library header, which defines the
`tcp_recv`, `tcp_sent`, `tcp_err`, `tcp_poll`, `tcp_connect`, and `tcp_accept` functions that take the
`tcp_recv_fn`, `tcp_sent_fn`, `tcp_err_fn`, `tcp_poll_fn`, `tcp_connected_fn`, and `tcp_accept_fn` function pointers (callback). Those function pointer typedefs have `err_t` in their signature.

You can verify this by checking the source. Under Linux, using the boards manager version of the esp8266 core, the header resides here:
`~/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h`

You can also see this from their lwIP's online documentation:
tcp_recv_fn docs:
http://www.nongnu.org/lwip/2_0_x/tcp_8h.html#a780cfac08b02c66948ab94ea974202e8
err_t docs:
http://www.nongnu.org/lwip/2_0_x/group__infrastructure__errors.html#gaf02d9da80fd66b4f986d2c53d7231ddb

whitebird, 2017-11-19:
@max5555 I've uploaded it here: 
ESPAsyncTCP.cpp: https://gist.github.com/whitebird/a376f47fdb3d6e7befaa4677e07c0d8c
ESPAsyncTCP.h: https://gist.github.com/whitebird/e9bb2637ee19f648318fd0716bbedd75

me-no-dev, 2017-11-19:
merged ;)

closed 2017-11-19 by me-no-dev