me-no-dev/ESPAsyncTCP #61: 2.3.0  ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'long int (*)...

ghislainborremans, 2017-10-19:
Hello , 
I can not compile due to the erros below:

ESPAsyncTCP\src\ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]
   tcp_accept(_pcb, &_s_accept);

any help would be appreciated. 

thanks in advance 

Ghislain

me-no-dev, 2017-10-19:
update your esp8266 arduino to the latest git :)

cwca02, 2017-10-19:
I'm having the same issue.  I updated esp8266 to the latest and I'm still having it.  Any help would be greatly appreciated.

ghislainborremans, 2017-10-19:
Updated but it stays the same. ( as reference: i am trying to run the FSBrowserNG UserConfigExample )

cwca02, 2017-10-20:
Go to c2d4c16.  I just implemented those changes and I'm golden.

ghislainborremans, 2017-10-20:
This is correct, the above solution works ( i copied the file contents from ESPAsyncTCP.cpp AND ESPAsyncTCP.h )
thans for the support.

closed 2017-10-20 by ghislainborremans