me-no-dev/ESPAsyncTCP #62: Build failed on Arduino 2.4.0-rc2

romansavrulin, 2017-10-23:
When updating to https://github.com/esp8266/Arduino/releases/tag/2.4.0-rc2

Build of the library fails with

```
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
     tcp_recv(_pcb, &_s_recv);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, uint16_t) {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
     tcp_sent(_pcb, &_s_sent);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, int8_t) {aka void (*)(void*, signed char)}' to 'tcp_err_fn {aka void (*)(void*, long int)}' [-fpermissive]
     tcp_err(_pcb, &_s_error);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*) {aka signed char (*)(void*, tcp_pcb*)}' to 'tcp_poll_fn {aka long int (*)(void*, tcp_pcb*)}' [-fpermissive]
     tcp_poll(_pcb, &_s_poll, 1);
                               ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, int8_t) {aka void (*)(void*, signed char)}' to 'tcp_err_fn {aka void (*)(void*, long int)}' [-fpermissive]
   tcp_err(pcb, &_s_error);
                         ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'AsyncClient& AsyncClient::operator=(const AsyncClient&)':
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:166:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
     tcp_recv(_pcb, &_s_recv);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:167:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, uint16_t) {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
     tcp_sent(_pcb, &_s_sent);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:168:28: error: invalid conversion from 'void (*)(void*, int8_t) {aka void (*)(void*, signed char)}' to 'tcp_err_fn {aka void (*)(void*, long int)}' [-fpermissive]
     tcp_err(_pcb, &_s_error);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:169:31: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*) {aka signed char (*)(void*, tcp_pcb*)}' to 'tcp_poll_fn {aka long int (*)(void*, tcp_pcb*)}' [-fpermissive]
     tcp_poll(_pcb, &_s_poll, 1);
                               ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
[... 9 lines cut ...]
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:291:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, uint16_t) {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
     tcp_sent(_pcb, &_s_sent);
                            ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
                  ^
/Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:292:31: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*) {aka signed char (*)(void*, tcp_pcb*)}' to 'tcp_poll_fn {aka long int (*)(void*, tcp_pcb*)}' [-fpermissive]
     tcp_poll(_pcb, &_s_poll, 1);
                               ^
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
[... cut ...]
In file included from /Users/rs/projects/SmartWatch/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:27:0:
/Users/rs/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]
 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;
                  ^
```

Not sure if this is issue of your library, but downgrade to rc1 works fine.

kv2000in, 2017-10-25:
Unable to compile with 2.3.0. Similar error--- 

Arduino: 1.6.5 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
Using library ESP8266WiFi in folder: /Users/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi 
Using library ESP8266mDNS in folder: /Users/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS (legacy)
Using library ArduinoOTA in folder: /Users/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ArduinoOTA 
Using library Hash in folder: /Users/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/Hash 
Using library ESP AsyncTCP in folder: /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master 
Using library ESP Async WebServer in folder: /Users/Documents/Arduino/libraries/ESPAsyncWebServer-master 

/Users/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
/Users/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
     tcp_recv(_pcb, &_s_recv);
                            ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
/Users/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
                  ^
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
     tcp_sent(_pcb, &_s_sent);
                            ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
/Users/kv2000in/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
                  ^
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]
     tcp_err(_pcb, &_s_error);
                            ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
/Users/kv2000in/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
                  ^
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]
     tcp_poll(_pcb, &_s_poll, 1);
                               ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
/Users/kv2000in/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
                  ^
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]
   tcp_err(pcb, &_s_error);
                         ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
/Users/kv2000in/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
                  ^
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp: In member function 'AsyncClient& AsyncClient::operator=(const AsyncClient&)':
/Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:166:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
     tcp_recv(_pcb, &_s_recv);
                            ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
[... cut ...]
   tcp_accept(_pcb, &_s_accept);
                              ^
In file included from /Users/kv2000in/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:27:0:
/Users/kv2000in/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]
 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;
                  ^
Error compiling.

kv2000in, 2017-10-25:
My bad - forgot to check other posts. This has already been taken care of 

https://github.com/me-no-dev/ESPAsyncTCP/commit/c2d4c168d41b78d9abd71a074c80eb8087c33561

andig, 2017-11-02:
Fixed in master and can be closed?

fdarling, 2017-12-18:
I believe this is a duplicate of #60 and it was fixed in https://github.com/me-no-dev/ESPAsyncTCP/commit/a57560d977602145cfadae84ae28f2ec69b361f5 so this can now be marked as closed.

kylegordon, 2017-12-29:
Is the issue I described at https://github.com/marvinroger/homie-esp8266/issues/444#issuecomment-354472834 an ESPAsyncTCP issue, or a homie-esp8266 problem?

Basically, with v3.5.0 of PlatformIO, ```pio run``` pulls in ESPAsyncTCP 1.0.1, which breaks, and ```pio lib update``` pulls in 1.1.3, which works.

closed 2019-10-05 by stale[bot]