me-no-dev/ESPAsyncTCP #63: incompatible with new SDK 2.1

universam1, 2017-10-23:
moved from me-no-dev/AsyncTCP/issues/5

I am unable to build with latest commits in Arduino for SDK 2.1

Please see the log output here:
https://travis-ci.org/universam1/iSpindel/builds/288929668

If the fixes are already committed, can you please bump the version so they are accessible via PlatformIO library update?
Thank you!

ploys, 2017-10-25:
Have the same problem on arduino IDE

E:\projects\electronic\arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

E:\projects\electronic\arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
     tcp_recv(_pcb, &_s_recv);
                            ^
In file included from E:\projects\electronic\arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:
C:\Users\nikolayba\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
                  ^

ploys, 2017-10-25:
Looks like AsyncClient use long as error type, but lwip use signed char for err_t.
Tried add #define LWIP_ERR_T s32_t to lwip cc.h, but in this case has similar type error in esp8266\2.3.0\libraries\ESP8266WiFi\src\WiFiClient.cpp:124:37 :)

romansavrulin, 2017-10-25:
see #62 and https://github.com/me-no-dev/ESPAsyncTCP/commit/c2d4c168d41b78d9abd71a074c80eb8087c33561

universam1, 2017-10-25:
@romansavrulin @me-no-dev where is the problem to bump the lib version to be accessible for PIO?

universam1, 2017-11-01:
@me-no-dev what should we do now? Any suggestions?

andig, 2017-11-01:
> Any suggestions?

You can use the git version of this library which contains the relevant fixes.

universam1, 2017-11-01:
here is a workaround for now: https://github.com/platformio/platform-espressif8266/issues/49#issuecomment-341142740

andig, 2017-11-02:
Fixed in master and can be closed?

egon0, 2017-11-06:
seems to me that the problem ist not fixed. using freshly checked out master-branch throws exact the same errors in arduino ide.

me-no-dev, 2017-11-06:
@egon0 something in your setup is the wrong version ;) code works fine under the new LwIP

egon0, 2017-11-06:
After manually applying c2d4c16 on top of master it works. Strange thing.

sreichholf, 2017-11-11:
it looks like  c2d4c16 has been force pushed away. It's not really in the tree anymore.
Probably by accident?!

andig, 2017-11-11:
There is https://github.com/me-no-dev/ESPAsyncTCP/commit/036ea44f8a3f5442a506fd2704f26335ba9c32ad now which seems to align the signatures.

fdarling, 2017-12-18:
I believe this is a duplicate of #60 and it was fixed in https://github.com/me-no-dev/ESPAsyncTCP/commit/a57560d977602145cfadae84ae28f2ec69b361f5 so this can now be marked as closed.

closed 2019-10-05 by stale[bot]