me-no-dev/ESPAsyncTCP #65: Client https (SSL)

mcmega, 2017-11-01:
Hi!
ESP8266 (ESP12)
Settings: Default
**How to start the client SSL?**
Dont work with:
http:// and https:// only www
Why?
```
#define OTA_HOST        "updates.soi-tech.com"
#define OTA_PATH        "/WalkingLight.txt"

bool otaAvailable() {
  if (WiFi.status() == WL_CONNECTED) {
//    AsyncClient client;
    SyncClient client;
    if (!client.connect(OTA_HOST, 80)) {
      Serial.printf("Connect Failed to: %s\n", OTA_HOST);
      return false;
    }
    client.setTimeout(3);
    if (client.printf(
          "GET %s HTTP/1.1\r\n"
          "Host: %s\r\n"
          "Connection: close\r\n"
          "\r\n", OTA_PATH, OTA_HOST) > 0) {
      while (client.connected() && client.available() == 0) {
        delay(1);
      }
      while (client.available()) {
        Serial.write(client.read());
      }
      if (client.connected()) {
        client.stop();
      }
    } else {
      client.stop();
      Serial.println("Send Failed");
      while (client.connected()) delay(0);
    }
    return true;
  } else return false;
}
```
Serial:
```
HTTP/1.1 200 OK␍␊
Server: nginx/1.10.3␍␊
Date: Wed, 01 Nov 2017 13:15:28 GMT␍␊
Content-Type: text/plain␍␊
Content-Length: 100␍␊
Last-Modified: Tue, 31 Oct 2017 02:00:02 GMT␍␊
Connection: close␍␊
ETag: "59f7d922-64"␍␊
Accept-Ranges: bytes␍␊
␍␊
1 F52187696F59EE5A1D85285D0F0B62A8 29131F8A6A853402AAB5433D521B1EC0 EA9BF8DE9BF008DFC94ECDE7260F4409
```
1 F52187696F59EE5A1D85285D0F0B62A8 29131F8A6A853402AAB5433D521B1EC0 EA9BF8DE9BF008DFC94ECDE7260F4409 - content of txt file.

## How to read the contents of a files *.txt, *.bin to const char?
### Where to specify a fingerprint and use https with SSL?
### Is it possible to use an asynchronous server (AsyncClient client)?

ps Don't work, if I use:
```
server.on("/ota/check", HTTP_GET, [](AsyncWebServerRequest * request) {
    if (otaAvailable()) {
      request->send(200, "text/plain; charset=utf-8", "true");
    } else {
      request->send(200, "text/plain; charset=utf-8", "false");
    }
});
```

mcmega, 2017-11-03:
Something I can not do anything ...
Tell me please, How to read only contents of a files *.txt, to const char *response = ...?

me-no-dev, 2017-11-03:
@mcmega sorry but you seem to take github issues as something that is dedicated to user issues. It is not ;) If you want help with your own problems using the lib, this is not the place. Please use Gitter chat for that (link in README). Closing this and most of your other "issues" as well.

closed 2017-11-03 by me-no-dev

mcmega, 2017-11-03:
Thanks for the answer, but I did not find a link to Gitter

me-no-dev, 2017-11-03:
[image]