me-no-dev/ESPAsyncTCP #66: Comiler errors after upgrade in VS Code/PlatformIO

chessweb01, 2017-11-01:
After updating PlatformIO in Visual Studio Code 1.17.2 I get the following errors when compiling my project (all originating in ESPAsyncTCP.cpp):

Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.o
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signe
d char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:318:1
8: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, uint16_t) {aka signed cha
r (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
tcp_sent(_pcb, &_s_sent);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:319:1
8: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, int8_t) {aka void (*)(void*, signed c
har)}' to 'tcp_err_fn {aka void (*)(void*, long int)}' [-fpermissive]
tcp_err(_pcb, &_s_error);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:321:1
8: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*) {aka signed char (*)(void
*, tcp_pcb*)}' to 'tcp_poll_fn {aka long int (*)(void*, tcp_pcb*)}' [-fpermissive]
tcp_poll(_pcb, &_s_poll, 1);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:320:1
8: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, int8_t) {aka void (*)(void*, signed
char)}' to 'tcp_err_fn {aka void (*)(void*, long int)}' [-fpermissive]
tcp_err(pcb, &_s_error);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:321:1
8: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'AsyncClient& AsyncClient::operator=(const AsyncClient&)':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:166:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka sign
ed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:318:1
8: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:167:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, uint16_t) {aka signed ch
ar (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
tcp_sent(_pcb, &_s_sent);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:319:1
8: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:168:28: error: invalid conversion from 'void (*)(void*, int8_t) {aka void (*)(void*, signed
[... 30 lines cut ...]
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:319:1
8: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:292:31: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*) {aka signed char (*)(voi
d*, tcp_pcb*)}' to 'tcp_poll_fn {aka long int (*)(void*, tcp_pcb*)}' [-fpermissive]
tcp_poll(_pcb, &_s_poll, 1);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:320:1
8: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
^
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, int8_t) {aka signed char
 (*)(void*, tcp_pcb*, signed char)}' to 'tcp_accept_fn {aka long int (*)(void*, tcp_pcb*, long int)}' [-fpermissive]
tcp_accept(_pcb, &_s_accept);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\users\rsel\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:317:1
8: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]
void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;
^

chessweb01, 2017-11-01:
This is related to issues #56, #57, #58. In order to fix it one has to add explicit type conversions  

    tcp_recv(_pcb, (tcp_recv_fn)&_s_recv);
    tcp_sent(_pcb, (tcp_sent_fn)&_s_sent);
    tcp_err(_pcb, (tcp_err_fn)&_s_error);
    tcp_poll(_pcb, (tcp_poll_fn)&_s_poll, 1);
    tcp_accept(_pcb, (tcp_accept_fn)&_s_poll, 1);

in file ESPAsyncTCP.cpp.

andig, 2017-11-02:
This has been fixed in master and can be closed?

chessweb01, 2017-11-02:
No, it is not fixed in master.

atanisoft, 2017-11-02:
this is due to the platformio espressif8266 version still being 1.4.0 and not the 2.4.0 that has the new prototypes. You can fix this by specifying this in your platformio.ini:

[env:esp8266]
platform=https://github.com/platformio/platform-espressif8266.git#feature/stage
lib_deps=ESPAsyncTCP@1.1.1

at least this is what worked for me.

chessweb01, 2017-11-02:
You are right, works for me, too.

andig, 2017-11-02:
Sorry, forgot to say that I‘m using feature/stage.

watershade, 2017-11-06:
I have meet the same problem. And thank for all yours'work.

closed 2017-11-06 by me-no-dev

watershade, 2017-11-07:
why close. The same problem for platformio, can not repair. I think you need repair it in new version.
1.1.1 is not work