me-no-dev/ESPAsyncTCP #68: version1.1.1 on platfromio still compile error

watershade, 2017-11-07:
The error message is : 
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {akasigned char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
tcp_sent(_pcb, &_s_sent);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]
tcp_err(_pcb, &_s_error);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]
tcp_poll(_pcb, &_s_poll, 1);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]
tcp_err(pcb, &_s_error);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'AsyncClient& AsyncClient::operator=(const AsyncClient&)':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:166:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:167:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
tcp_sent(_pcb, &_s_sent);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:168:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]
tcp_err(_pcb, &_s_error);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]
void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:169:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]
tcp_poll(_pcb, &_s_poll, 1);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'long int AsyncClient::_connected(void*, long int)':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:290:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:291:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]
tcp_sent(_pcb, &_s_sent);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:292:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]
tcp_poll(_pcb, &_s_poll, 1);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
^
Compiling .pioenvs/nodemcuv2/lib/ESP8266mDNS/ESP8266mDNS.o
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]
tcp_accept(_pcb, &_s_accept);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/home/galileo/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]
void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;
^
Compiling .pioenvs/nodemcuv2/lib/ArduinoOTA/ArduinoOTA.o
*** [.pioenvs/nodemcuv2/lib/ESPAsyncTCP_ID305/ESPAsyncTCP.o] Error 1

atanisoft, 2017-11-07:
Please see the previous issue filed for this that you commented on. You must upgrade to the stage version of esp8266 for platformio and use version 1.1.1 to get this working. I posted the exact pieces you need in the platformio.ini file in the other issue.

atanisoft, 2017-11-07:
Issue #66 has the fix.

watershade, 2017-11-08:
Thanks, It is done

closed 2017-11-08 by watershade