me-no-dev/ESPAsyncTCP #80: Compile fails with SSL enabled

AcuarioCat, 2018-01-14:
Hi,
when I compile with ASYNC_TCP_SSL_ENABLED defined I'm getting a load of errors (without it is fine):
SyncClient.cpp.o*: In function SyncClient::_attachCallbacks_Disconnect()
SyncClient.cpp:209: undefined reference to AsyncClient  AsyncClient(tcp_pcb*, SSL_CTX_*)
SyncClient.cpp:209: undefined reference to AsyncClient  connect(IPAddress, unsigned short, bool)

Error linking for board NodeMCU 1.0 (ESP-12E Module)
 
SyncClient.cpp.o*: In function SyncClient::connect(IPAddress, unsigned short, bool)
Build failed for project 'TestSSL'
SyncClient.cpp:71: undefined reference to AsyncClient  AsyncClient(tcp_pcb*, SSL_CTX_*)
SyncClient.cpp:64: undefined reference to AsyncClient  connect(IPAddress, unsigned short, bool)
 
SyncClient.cpp.o*: In function SyncClient::connect(IPAddress, unsigned short)
SyncClient.cpp:76: undefined reference to AsyncClient  connect(char const*, unsigned short, bool)
 
SyncClient.cpp.o*: In function SyncClient::connect(char const*, unsigned short, bool)
SyncClient.cpp:94: undefined reference to AsyncClient  AsyncClient(tcp_pcb*, SSL_CTX_*)
 
SyncClient.cpp.o*: In function _M_init_functor
functional:1987: undefined reference to AsyncClient  connect(char const*, unsigned short, bool)
 
collect2.exe*: error: ld returned 1 exit status

Any ideas? I'm using the latest Arduino build.

adrionics, 2018-05-21:
Did you ever get this working?

beejay31, 2018-06-14:
I'm having a similar same issue, both with the SyncClient.h and the ESPAsyncClient.h lib files.  I'm on ESP8266 2.4.1 updated from Github.  I've modified the SyncClient Example to test SSL connection:

#define ASYNC_TCP_SSL_ENABLED true

#include <SyncClient.h>
#include <ESP8266WiFi.h>

const char* ssid = "****************";
const char* password = "*****************";

void setup(){
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }
  Serial.printf("WiFi Connected!\n");
  Serial.println(WiFi.localIP());
  
  SyncClient client;
  if(!client.connect("www.google.com", 443, true)){
    Serial.println("Connect Failed");
    return;
  }
  client.setTimeout(2);
  if(client.printf("GET / HTTP/1.1\r\nHost: www.google.com\r\nConnection: close\r\n\r\n") > 0){
    while(client.connected() && client.available() == 0){
      delay(1);
    }
    while(client.available()){
      Serial.write(client.read());
    }
    if(client.connected()){
      client.stop();
    }
  } else {
    client.stop();
    Serial.println("Send Failed");
    while(client.connected()) delay(0);
  }
}

void loop(){
  delay(0);
}

The compiler returns the following errors:

C:\Users\BRENDA~1\AppData\Local\Temp\arduino_build_514642\sketch\sketch_jun13a.ino.cpp.o:(.text.setup+0x2c): undefined reference to `SyncClient::connect(char const*, unsigned short, bool)'

C:\Users\BRENDA~1\AppData\Local\Temp\arduino_build_514642\sketch\sketch_jun13a.ino.cpp.o: In function `setup':

C:\Users\BrendanRoss\Documents\Arduino\sketch_jun13a/sketch_jun13a.ino:9: undefined reference to `SyncClient::connect(char const*, unsigned short, bool)'

I modified the SyncClient example with some ASyncClient code and I get the same undefined reference error for the same connection definition.

Any clues as to why?

closed 2019-10-05 by stale[bot]

minkosimon, 2022-06-15:
Hello this issue is still open ? I get the same issue after defining MACRO ASYNC_TCP_SSL_ENABLED