me-no-dev/ESPAsyncTCP #82: Wont compile - rename tcp_axtls.c to .cpp?

gjt211, 2018-01-24:
Using Arduino IDE.

I have been successfully using 2.4.0 rc-1 core and all the Async libs from about October last year.

I recently updated core to 2.4.0 release and also updated the Async libs but now my code won't compile.
The error I get is
```
Arduino/libraries/ESPAsyncTCP-master/src/tcp_axtls.h:87:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp);
```
I have been searching (and trying many various suggestions) for three weeks now for something that may help.
One of the comments I found somewhere was .c does not have a `bool` type and a suggestion was to rename the `.c` to a `.cpp` file - I tried this and the code now compiles but won't connect to my server.
My code has not changed, and using the older core and libs compiles, and does everything it should. With the newer core/libs it now compiles but won't connect.

Using the AsyncMqttClient example, I added code for all the possible error types. The error I see is
```
AsyncMqttClientDisconnectReason::TCP_DISCONNECTED
```
and my mqtt broker log file shows every-time a connection is attempted (ip address not real)
```
New connection from 111.11.11.11 on port 8883.
OpenSSL Error: error:140780E5:SSL routines:SSL23_READ:ssl handshake failure
Socket error on client <unknown>, disconnecting.
```
My fingerprint has not changed as code compiled with earlier libs works fine.

It seems that AsyncTCP is not connecting using TLS, as it always returns 0.
I have tried this on 3 different PCs (One MacMini, One MackBook Pro, One Windows10)

Question 1. Why can't I compile without changing `tcp_axtls.c` to `tcpaxtls.cpp`?
Question 2. Will changing the extension to `.cpp` cause other problems (like the problem I have)?

tarzan115, 2018-01-24:
`#include <stdbool.h>` to `tcp_axtls.c` and it will compile

gjt211, 2018-01-24:
Hi @tarzan115, thank you for that - it now compiles and works!!!
One of the 'fixes' I tried was almost exactly that, but I put it at the top of my sketch without success of course!

Is there a reason why this is needed? Am I the only one having this problem, and if so my question would be why? What is wrong with the three PCs I use?

Thanks again!

tarzan115, 2018-01-24:
No, I got the same problem like you before. I've done that too. So I think maybe `tcp_axtls.h` called somewhere before it calls in your sketch.

closed 2018-01-25 by gjt211

dejanra, 2018-04-08:
Where I need change this '#include <stdbool.h> to tcp_axtls.c'? In 'ESPAsyncTCP-master'? I can't find it.

dejanra, 2018-04-08:
Where I need change '#include <stdbool.h> to tcp_axtls.c'? In 'ESPAsyncTCP-master'? I can't find it. I changed tcp_axtls.c to .cpp and compile, but Wemos D1 wdt reboot. help!

tarzan115, 2018-04-09:
here: https://github.com/me-no-dev/ESPAsyncTCP/pull/84/files

dejanra, 2018-04-09:
What exactly need to change in ESPAsyncTCP-master (or async-mqtt-client-master)? I changed #include <stdbool.h>  (tcp_axtls.c) in 'src/tcp_axtls.h' and fd_data->tcp_pbuf = NULL; in 'src/tcp_axtls.c' but I got error:  void tcp_ssl_err(struct tcp_pcb *tcp, tcp_ssl_error_cb_t arg);

I am using core 2.4.1 and latest library ESPAsyncTCP-master and async-mqtt-client-master.

tarzan115, 2018-04-10:
you just need to change only in `src/tcp_axtls.h` file of `ESPAsyncTCP-master`.

dejanra, 2018-04-10:
But, if I change only in src/tcp_axtls.h - Add #include <stdbool.h> and change to tcp_axtls.c it compile, but TLS/SSL not working. 
If comment //#if ASYNC_TCP_SSL_ENABLED, then I get 

exit status 1
'class AsyncMqttClient' has no member named 'setSecure'

tarzan115, 2018-04-13:
in cmd in windows `py .\get-fingerprint.py --host CloudMQTT.com --port 8883` (make sure you had installed python)

dejanra, 2018-04-13:
I was successfully generate fingerprint and change #define ASYNC_TCP_SSL_ENABLED 1 and #include <stdbool.h>.
Normally compiling and CloudMQTT says that Wemos D1 is connected over SSL, but Wemos stuck here:

Connected to Wi-Fi.
Connecting to MQTT...

and after some minutes wdt reset 

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40034320 depc=0x00000000

ctx: sys 
sp: 3ffffc60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe00:  3ffeda78 40102136 3fff0a68 402153c4  
3ffffe10:  3ffef52c 000006f3 000006f3 4010020c  
3ffffe20:  00000700 00000000 3fff430c 40100690  
3ffffe30:  3ffffe40 000006f3 000006f3 402163fc  
3ffffe40:  3ffffe70 3fff430c 3fff1834 4020f5d6  
3ffffe50:  40100296 00000023 3fff430c 40211f58  
3ffffe60:  fffffffd 3fff430c 3fff15dc 40201315  
3ffffe70:  3ffed2f0 00000006 00000000 000006f3  
3ffffe80:  00000006 3ffef52c 3fff109c 00000030  
3ffffe90:  3ffeeb70 3ffed188 00000002 00000001  
3ffffea0:  3ffef168 3fff430c 3fff430c 40206463  
3ffffeb0:  3fff178c 3fff08c0 3fff430c 4021351a  
3ffffec0:  00000006 00000001 24221f1c 3fff08dc  
3ffffed0:  00000000 3fff430c 3fff178c 4020648c  
3ffffee0:  3fff0a68 3fff0b50 3fff4324 4020fd00  
3ffffef0:  3fff178c 3fff08c0 3fff4324 402118ae  
3fffff00:  660da8c0 00000000 3ffeae62 00000000  
3fffff10:  00000000 00000036 3fffff80 3fff4342  
3fffff20:  3fff0a68 3fff4324 3fff0b50 40215b69  
3fffff30:  00000014 000006f3 000006f3 3fff0a68  
3fffff40:  3fffdc80 3fff1014 3fff430c 3fff109c  
3fffff50:  00000008 3fff0a68 3fff4324 4020f481  
3fffff60:  3fffdc80 3fff1014 3fff430c 4010453c  
3fffff70:  4023825e 3fff1014 3fff430c 40238270  
3fffff80:  3fff4334 3fff4324 00000002 00000000  
3fffff90:  4023302a 00000000 3fff430c 4023a1df  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

tarzan115, 2018-04-13:
use this tool https://github.com/me-no-dev/EspExceptionDecoder to decode your stack. and give me a slice of your code, it will better to help

tarzan115, 2018-04-14:
I didn't see anything wrong with your code. please reinstall `EspExceptionDecode` tool follow by https://github.com/me-no-dev/EspExceptionDecoder#installation to decode your stack.

dejanra, 2018-04-14:
I can't install EspExceptionDecoder. I tryed in IDE 1.8.5, 1.8.1 and 1.6.5 but nothing happened, I don't see tool in Arduino IDE.
I tryed EspExceptionDecoder-master in Arduino/tools, and make folders Arduino/tools/EspExceptionDecoder/tool/EspExceptionDecoder.jar

dejanra, 2018-04-19:
Exception decoder:

Decoding 23 results
0x402165a4: raw_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/raw.c line 498
0x4010020c: _umm_free at C:\Users\dejan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\dejan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c line 1739
0x402163fc: mem_free at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 152
0x4020f5d6: memp_free at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x40211f58: pbuf_free_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306
0x40201315: tcp_ssl_read at C:\Users\dejan\Documents\Arduino\libraries\ESPAsyncTCP-master\src/tcp_axtls.c line 477
0x40215db0: ip4_output_if_opt at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 788
0x40100866: ppProcessTxQ at ?? line ?
0x40206463: AsyncClient::_recv(tcp_pcb*, pbuf*, long) at C:\Users\dejan\Documents\Arduino\libraries\ESPAsyncTCP-master\src/ESPAsyncTCP.cpp line 665
0x4021351a: tcp_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c line 1123
0x4020648c: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at C:\Users\dejan\Documents\Arduino\libraries\ESPAsyncTCP-master\src/ESPAsyncTCP.cpp line 665
0x4020fd00: tcp_process_refused_data at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c line 1779
0x402118ae: tcp_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c line 357
0x40215b69: ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 685
0x4020f481: ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 182
0x4010453c: esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 433
0x4023825e: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 357
0x40238270: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 365
0x4023302a: ppPeocessRxPktHdr at ?? line ?
0x4023a1df: ets_snprintf at ?? line ?

Help! :)

dejanra, 2018-04-19:
[image]

tarzan115, 2018-04-20:
try this PR https://github.com/me-no-dev/ESPAsyncTCP/pull/84 to fix that error ;)

dejanra, 2018-04-20:
It does not working! 

Which Arduino IDE and ESP8266 core version are You use? It is logically that newest version of library is the best.

dejanra, 2018-04-22:
The problem is with ESP Arduino Core 2.4.1. Authors please fix that in version 2.4.2 :)

With 2.4.0 rc1 works like a charm!