me-no-dev/ESPAsyncTCP #89: Optimizing for real time control if using FreeRTOS in main program

terryjmyers, 2018-03-12:
The WebServer and WebSockets portion of my project are a luxury in my project and therefore should generally have a lower priority than the default that I see in the library.

AsyncTCP.cpp is where the FreeRTOS task is created using "xTaskCreatePinnedToCore", pinned to core 1, with a fairly high priority.  Is there any reason why I can't change it to "xTaskCreate" to allow the scheduler to decide and also give it a lower priority, the same priority as my "housekeeping" task?

Along the same lines the "_async_service_task" allows a 1ms "delay" between servicing requests on the queue.  Is there any reason I can't make this longer say 10 or even 100ms?

Besides increasing TCP/IP response times, is there any reason you can see from a system stability standpoint why this would be bad?  Did you choose these settings because most people don't even use (or know about) core 1, and since I will be allowing the task scheduler to schedule on either core, this would nullify that design decision?

terryjmyers, 2018-03-17:
OK, so I've run some tests.  I modified the function _start_async_task() in AsyncTCP.cpp to be:
`if(!_async_service_task_handle){
			xTaskCreate(
				_async_service_task,//pvTaskCode
				"async_tcp",//pcName
				8192,//usStackDepth
				NULL,//pvParameters
				1,//uxPriority
				&_async_service_task_handle//pxCreatedTask
			);		`

This modification was stable and I verified that it ran on different cores by adding a Serial.println(xPortGetCoreID()); right before the packet is handled (_handle_async_event(packet);).

However I tried modifying the vTaskDelay(1) to 100, 10, 5, 3, and 2 and all were unstable if I spammed refresh on the SPIFFSEditor "\edit" page.  Even vTaskDelay(1) I was able to get the ESP32 to crash if I spammed hard enough.  I guess I was filling up the heap faster than the ESP32 could serve pages.

So I'll at least have the Async library run on multiple cores with a lower priority.  I'll need to try spamming websockets and telnet as well as another stability test

RudyFiero, 2018-03-18:
Based on what you wrote here I think you are in the wrong library. My understanding was that this is for ESP8266. What I think you want is the library for the ESP32. 
https://github.com/me-no-dev/AsyncTCP

closed 2019-10-05 by stale[bot]