me-no-dev/ESPAsyncTCP #108: ESP8266 TCP Source Port Uses Fixed Sequence

brandond, 2019-02-22:
There is an issue with the esp8266 lwip library where the tcp_new() function uses a fixed sequence of source ports every time. This means that a node that is rebooting frequently and reconnecting to the same server may experience issues due to the host perceiving the packets as being retransmissions of old traffic - they have the exact same source, destination, and initial sequence number.

This has been reported and discussed in the Arduino core in https://github.com/esp8266/Arduino/issues/632; their fix was to allow the user to call `setLocalPortStart()`, which sets an initial source port that is used (and incremented) to override `local_port` within the `tcp_pcb` returned by `tcp_new()`.

Does ESPAsyncTCP have a work-around for this issue, and if not, could one be added? I believe I'm experiencing occasional dropped traffic caused by this oddity.

mcspr, 2019-03-29:
@brandond just so you're aware, most recent Core version has a different approach for this. see esp8266/Arduino#5906

brandond, 2019-09-22:
So is this fixed, or just fixed in a different version of the lib?

mcspr, 2019-09-26:
Should be fixed upstream, Core >=2.5.2 with lwip2 selected will randomize ports by default

closed 2019-12-09 by stale[bot]