me-no-dev/ESPAsyncTCP #112: Wifi scan network issues for webpage

fu-techiot, 2019-06-19:
Hey, Hello There,

I have one sophisticated configuration web page hosted on esp8266's AP, using this I can easily set wifi networks, web services like (HTTP, MQTT, TCP, etc), access point name, password, access point IP(192.168.4.1)and other details too.

I'm sending one command for wifi scanning, it will scan available wifi networks, and give all the scanned network at webpage so that I can use one of them. 
but the problem is esp8266 is giving network scanned result "0".  it is not able to scan available networks. 
-->**there is no hardware issue.** 
-->it is **working fine with the default code available at esp8266-->wifiscan.ino**
-->it is also working fine at the start of the setup function. but when I'm sending a command from the webpage it gives "No Networks Found" 
---> it's **working just fine for ESP32** with "ESPAsyncWebServer.h" and AsyncTCP.h library 

but no luck with esp8266 !!!
Thanks,
i'm using this code for testing, u can check this out.

```
> #include <ESP8266WiFi.h>
> #include "ESPAsyncWebServer.h"
> 
> const char *ssid = "esp8266_thing";
> const char *password = "1234567890";
> 
> int LED1 = D5 ;

> String Page = "<!DOCTYPE html><html><head><h2>OPEN CORNER WEB SERVER DEMO </h2></head><body><h1 style=\"color: blue;font-family:verdana;font-size:300%;\">DEMO SEMINAR 1</h1><p><a href=\"LEDOn\"><button style=\"color: green;font-family:verdana;padding: 50px 75px;font-size:300%\">ON</button></a>&nbsp;<a href=\"LEDOff\"><button style=\"color: red;font-family:verdana;padding: 50px 75px;font-size:300%\" >OFF</button></a></p></body></html>";
> 
> AsyncWebServer server(80);
> 
> void setup() {
> 
>   Serial.begin(115200);
> 
>   // Set WiFi to station mode and disconnect from an AP if it was previously connected
>   WiFi.mode(WIFI_STA);
>   WiFi.disconnect();
>   delay(100);
> 
>   Serial.println("Setup done");
>   
>   pinMode(LED1, OUTPUT);
>   WiFi.softAP(ssid, password);
> 
>   Serial.println();
>   Serial.print("IP address: ");
>   Serial.println(WiFi.softAPIP());
> Serial.println("scan start");
> 
>   // WiFi.scanNetworks will return the number of networks found
>   int n = WiFi.scanNetworks();
>   Serial.println("scan done");
>   if (n == 0) {
>     Serial.println("no networks found");
>   } else {
>     Serial.print(n);
>     Serial.println(" networks found");
>     for (int i = 0; i < n; ++i) {
>       // Print SSID and RSSI for each network found
>       Serial.print(i + 1);
>       Serial.print(": ");
>       Serial.print(WiFi.SSID(i));
>       Serial.print(" (");
>       Serial.print(WiFi.RSSI(i));
>       Serial.print(")");
>       Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : "*");
>       delay(10);
>     }
>   }
>   server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
> 
>     request->send(200, "text/html", Page );
>   });
> 
>   server.on("/LEDOn", HTTP_GET, [](AsyncWebServerRequest * request) {
> 
>     Serial.println("LED ON");
>     digitalWrite(LED1, HIGH);
> 
> Serial.println("scan start");
> 
>   // WiFi.scanNetworks will return the number of networks found
>   int n = WiFi.scanNetworks();
>   Serial.println("scan done");
>   if (n == 0) {
>     Serial.println("no networks found");
>   } else {
>     Serial.print(n);
>     Serial.println(" networks found");
>     for (int i = 0; i < n; ++i) {
>       // Print SSID and RSSI for each network found
>       Serial.print(i + 1);
>       Serial.print(": ");
>       Serial.print(WiFi.SSID(i));
>       Serial.print(" (");
>       Serial.print(WiFi.RSSI(i));
>       Serial.print(")");
>       Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : "*");
>       delay(10);
>     }
>   }
>     
>     request->send(200, "text/html", Page );
> 
>   });
> 
>   server.on("/LEDOff", HTTP_GET, [](AsyncWebServerRequest * request) {
> 
>     Serial.println("LED OFF");
>     digitalWrite(LED1, LOW)
>     request->send(200, "text/html", Page );
>   });
>   server.begin();
> }
> void loop()
> {
> 
> }

```

RudyFiero, 2019-06-19:
If someone wanted to test this with your supplied code, it would make things easier if you removed the > at the start of each line.

fu-techiot, 2019-06-21:
@RudyFiero okay, I did this thing cause in this testing .ino code there is a webpage code and in browser its showing HTML in a web form. and it's looking bit confusing, 

and may generate more errors, 
okey u can download the fulll code from here:
https://github.com/fu-techiot/testing-and-troubleshooting/blob/master/network%20scan%20error%20on%20webpage%20command.ino

closed 2019-10-05 by stale[bot]