me-no-dev/ESPAsyncTCP #130: ASyncTCP very slow Client/Server

notsolowki, 2019-10-13:
Why when i try to send a reply faster than 2 seconds it goes so slow. If i try 1 second if works a few times then locks up for 5-10 seconds and continues then repeats???

me-no-dev, 2019-10-14:
code? device? logs?

notsolowki, 2019-10-14:
Here is the code i used. both the client/server example and this code face the same issue,
`#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>

extern "C" {
#include <osapi.h>
#include <os_type.h>
}

#include "config.h"
#define SSID "ESP-TEST"
#define PASSWORD "123456789"
#define SERVER_HOST_NAME "esp_server"
#define TCP_PORT 7050
#define DNS_PORT 53
unsigned long now =0;
struct sampleStruct {
  int var1 = 253;
  //byte Array[20];
  float var2 = 5.5;
  unsigned long var3 = 999;
  unsigned long var4 = 1;
  unsigned long var5 = 9999;
  unsigned long var6 = 7777;
  unsigned long var7 = 6666;
  unsigned long var8 = 5555;
  // bool var9;
};
sampleStruct st;
AsyncClient* client = new AsyncClient;
static os_timer_t intervalTimer;

static void replyToServer(void* arg) {
  AsyncClient* client = reinterpret_cast<AsyncClient*>(arg);
  // send reply

  
}

/* event callbacks */
static void handleData(void* arg, AsyncClient* client, void *data, size_t len) {
  Serial.printf("\n data received from %s \n", client->remoteIP().toString().c_str());
 // Serial.write((uint8_t*)data, len);
  //os_timer_arm(&intervalTimer, 500, true); // schedule for reply to server at next 2s
}
void onConnect(void* arg, AsyncClient* client) {
  Serial.printf("\n client has been connected to %s on port %d \n", SERVER_HOST_NAME, TCP_PORT);
  //replyToServer(client);
}
void setup() {
  Serial.begin(115200);
  delay(20);
  WiFi.mode(WIFI_STA);
  WiFi.begin(SSID, PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(500);
  }

  client->onData(&handleData, client);
  client->onConnect(&onConnect, client);
  client->connect(SERVER_HOST_NAME, TCP_PORT);

//  os_timer_disarm(&intervalTimer);
//  os_timer_setfn(&intervalTimer, &replyToServer, client);
//  os_timer_arm(&intervalTimer, 500, true);

}

void loop() {
    if (millis() - now >= 500) {
        if (client->space() > 32 && client->canSend()) {
    client->add("NODE03", 7); //add verification1
   client->add((char *)&st, sizeof(sampleStruct));
    client->add("NODE03", 7); //add verification2
    client->send();
      Serial.println("sent");
   //client->free();
        }
  
 // replyToServer(client);
  now = millis();
    }
}

void testFunction(  void* arg, AsyncClient* client){
  if (millis() - now >= 500) {

   // AsyncClient* client = (AsyncClient);
    //replyToServer(void);
    
  }
}`

closed 2019-12-27 by stale[bot]