me-no-dev/ESPAsyncTCP #143: Don't use arduino String class

computski, 2020-04-20:
FYI.  Tried using ESPAsyncTCP along with a lot of string manipulation.  Saw a lot of hardware WDT timeouts and could not pin down exactly why/when.

I removed all use of the Arduino String class.  Instead used char arrays, memset, strcpy, strstr and pointers.   It now works reliably, no WDT resets.

Difficult to say whether the String class is incompatible with ESPAsyncTCP or whether it was causing the WDT resets on its own.  Avoid using it!

shuai132, 2020-05-18:
I use a lot of bit and complex datastructure(eg: std::string std::vector) in my project.
It works well!

CelliesProjects, 2020-05-20:
I noticed too that String should be avoided in callbacks.
In soms cases it works at first but fails after creating or manipulating a number of Strings.

sparkplug23, 2020-05-26:
Is there use of strings within asynctcp? Or simply you passed strings to it?

Yes, strings should be avoided at all costs. Memory leaks.

computski, 2020-05-30:
FYI, when receiving incoming TCP data, I was using;
static void handleData(void* arg, AsyncClient* client, void *data, size_t len) {
	String s = (char*)data;
..do stuff

and when transmitting over TCP I was using;

if (client->space() > 100 && client->canSend()) {
		char reply[100];
		s.toCharArray(reply, 100);
		client->add(reply, strlen(reply));
		client->send();
s is an Arduino String.  In principle even a String longer than 100 should be truncated rather than causing a buffer overrun. 

As mentioned, I saw sporadic WDT crashes from this module.  This is odd, given I had other modules in the project which did use String with no issues.   Anyway, the fix was to remove all Strings from the module using AsyncTCP and instead use cstring.  works fine, no crashes.

mhightower83, 2020-05-30:
@computski I cannot spot a problem in what you posted; however, if you would like to get a stack trace to debug your Hardware WDT crash. There is one file you can drop into your sketch folder to generate one. You need [`libraries/esp8266/examples/HwdtStackDump/hwdt_app_entry.cpp`](https://github.com/esp8266/Arduino/blob/1264394175631453e1269cf02dc4f70195335fbb/libraries/esp8266/examples/HwdtStackDump/hwdt_app_entry.cpp) from PR [Hardware WDT Stack Dump Tool #7010](https://github.com/esp8266/Arduino/pull/7010). The default options should be good. It uses the serial port set to 115200 by default.

The tool will dump both SYS and CONT stacks at reboot from an HWDT. You will need to run the _ESP8266 Exception Decoder_ on each stack. Since we don't have an execution address for the time of the crash, you have to look at the decoded result to try and get an idea of what might have been executing at the time of the crash. I know it is not easy, but at the least, it is more information than you got with just the `ets Jan 8 2013,rst cause:4, boot mode: (3,7)` message.

sparkplug23, 2020-05-30:
> FYI, when receiving incoming TCP data, I was using;
> static void handleData(void* arg, AsyncClient* client, void _data, size_t len) { String s = (char_)data;
> ..do stuff
>
> and when transmitting over [...]

My guess is the function is being used as a callback, and is creating Strings in memory but never releasing them. In general, String class on embedded is used to make it easier for novice programmers, but if you have the skill to use cstrings you absolutely should.

closed 2020-08-12 by stale[bot]