me-no-dev/ESPAsyncTCP #149: "Recommended" libraries that use ESPAsyncTCP might need reviewing

philbowles, 2020-07-01:
I have long been a user of ESPAsyncWebserver, ESPAsyncTCP and AsyncTCP and I trust those libraries, as I do all of your work.

I was very disappointed then when I changed all my firmware from PubSubClient to AsyncMqttClient (purely because it was "recommended" by *you* - an author I trust) because then my nightmare began: *all* of my firmware instantly broke. AsyncMqttClient is so full of bugs and problems, it is ***not fit for purpose***. It is my own fault for not evaluating it fully - I foolishly assumed that since you named it on your site that it was up to the same standard as your own libraries. Oh boy, was I wrong!

Rather than list all the bugs here with the evidence, I suggest you read [this document](https://github.com/philbowles/PangolinMQTT/blob/master/docs/bugs.md) and after having read it, you may want to reconsider including it on the list of libraries using *your* excellent offerings, as "People are judged by the company they keep" :)

kleini, 2020-07-01:
Will have a look, whether homieiot/homie-esp8266 can be ported to PangolinMQTT. I encouter some transmission issues with MQTT and need to use fixed versions of ESPAsyncTCP and AsyncMqttClient.

philbowles, 2020-07-01:
@kleini I would be interested in the "fix" to ESPAsyncTCP - Pangolin uses 100% bog-standard version with no problems. I suspect the need for your fixes a probably due to the problems in "The other library".

kleini, 2020-07-01:
https://github.com/me-no-dev/ESPAsyncTCP/pull/129

philbowles, 2020-07-01:
@kleini OK. Pangolin  doesnt have TLS / SSL etc in yet so I won't have spotted that

kleini, 2020-07-01:
No SSL support? That is a big problem especially with MQTT and WiFi as clear text passwords are transferred.

philbowles, 2020-07-01:
@kleini Yes, I appreciate that. Wanted to get it working first. Will be adding TLS very soon

closed 2020-09-14 by stale[bot]