me-no-dev/ESPAsyncTCP #156: compile error with #define ASYNC_TCP_SSL_ENABLED 1

Bounderx, 2020-10-23:
HI,

i have problems with compiling

`F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:764:6: warning: unused parameter 'tcp' [-Wunused-parameter]
 void AsyncClient::_s_data(void *arg, struct tcp_pcb *tcp, uint8_t * data, size_t len){
      ^
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:770:6: warning: unused parameter 'tcp' [-Wunused-parameter]
 void AsyncClient::_s_handshake(void *arg, struct tcp_pcb *tcp, SSL *ssl){
      ^
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:770:6: warning: unused parameter 'ssl' [-Wunused-parameter]
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:777:6: warning: unused parameter 'tcp' [-Wunused-parameter]
 void AsyncClient::_s_ssl_error(void *arg, struct tcp_pcb *tcp, int8_t err){
      ^
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:1254:57: warning: unused parameter 'arg' [-Wunused-parameter]
           c->onConnect([this](void * arg, AsyncClient *c){
                                                         ^
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:1320:53: warning: unused parameter 'arg' [-Wunused-parameter]
       c->onConnect([this](void * arg, AsyncClient *c){
                                                     ^
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: In member function 'err_t AsyncServer::_poll(tcp_pcb*)':
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:1325:10: error: 'errorTracker' was not declared in this scope
 c->_recv(errorTracker, pcb, p->pb, 0); //geändert von c->_recv(pcb, p->pb, 0);
          ^
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: At global scope:
F:\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:1333:7: warning: unused parameter 'err' [-Wunused-parameter]
 err_t AsyncServer::_recv(struct tcp_pcb *pcb, struct pbuf *pb, err_t err){
       ^

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

`

what is the problem ?

ohyeaah, 2020-10-24:
Hello @Bounderx ,
I confirm that I have the same issue as you.
7 of your messages are warnings about unused parameters and can safely be ignored.
The only error/problem is in file ESPAsyncTCP.cpp at line [1324](https://github.com/me-no-dev/ESPAsyncTCP/blob/master/src/ESPAsyncTCP.cpp#L1324).

Maybe someone can fix it?

philbowles, 2020-11-06:
Try this fork of mine where I have patched that bug: [patched lib](https://github.com/philbowles/ESPAsyncTCP)

ohyeaah, 2020-11-07:
> Try this fork of mine where I have patched that bug: [patched lib](https://github.com/philbowles/ESPAsyncTCP)

Thank you.
It's sad no one updates this repository (which is #1 on Google).

closed 2021-01-23 by stale[bot]