me-no-dev/ESPAsyncTCP #163: AsyncTCP + WiFiUDP -> crash

Miq1, 2021-02-18:
I am trying to run a Telnet server on an ESP8285 that is based on ``WiFiUDP.h``. Simultaneously on the same device a Modbus server is running that is built on ``ESPAsyncTCP``'s ``AsyncServer``.

As soon as a Telnet server function is used inside the AsyncTCP callback handler functions, the ESP8285 will reboot. Using Telnet functions outside the Async server code is working, as are the Async server functions without Telnet calls.

So it looks like the ``WiFiUDP`` innards are intertwined with those of ``AsyncTCP``.

Is there any known way to get around this?

Pablo2048, 2021-02-18:
Sync functions are not allowed to be used in async callbacks IMO. You can try to use https://github.com/me-no-dev/ESPAsyncUDP instead of WiFiUDP...

Miq1, 2021-02-18:
> 
> 
> Sync functions are not allowed to be used in async callbacks IMO. You can try to use https://github.com/me-no-dev/ESPAsyncUDP instead of WiFiUDP...

Well, that unfortunately would require to rewrite the Telnet server. I am using it in different sync contexts, so in the end I would have to have two implementations. If there is no other remedy, I must do that, but I rather would like an alternative.

Pablo2048, 2021-02-18:
Ok, so another alternative(s) I see are:
1. use buffer to copy content and flag to mark handling
2. use some sort of queue - fill it in async context, handle it in sync and vice versa

Miq1, 2021-02-18:
> 
> 
> Ok, so another alternative(s) I see are:
> 
>     1. use buffer to copy content and flag to mark handling
> 
>     2. use some sort of queue - fill it in async context, handle it in sync and vice versa

Thank you! I will contemplate over it a bit 😉

closed 2021-02-18 by Miq1