me-no-dev/ESPAsyncTCP #170: ESPAsyncTCP/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory

bastianhjaeger, 2021-07-09:
Missing ssl.h file. Where should I install this from? Whare shoud it be?

```
*** [.pio/build/my_app_debug/src/MqttHandler.cpp.o] Error 1
In file included from .pio/libdeps/my_app_debug/AsyncMqttClient/src/AsyncMqttClient.hpp:18,
                 from .pio/libdeps/my_app_debug/AsyncMqttClient/src/AsyncMqttClient.h:4,
                 from src/MqttHandler.h:16
.pio/libdeps/my_app_debug/ESPAsyncTCP/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory
   44 | #include "include/ssl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```
```
user@ubuntu:~/dev/my_app (devel *)$ pio lib update
Library Storage: /home/user/dev/my_app/.pio/libdeps/my_app
Updating me-no-dev/ESPAsyncTCP                1.2.2                              [Up-to-date]
Updating bblanchon/ArduinoJson                6.18.1                             [Up-to-date]
Updating marvinroger/AsyncMqttClient          0.8.2 @ ^0.8.2                     [Up-to-date]

Library Storage: /home/user/dev/my_app/.pio/libdeps/my_app_debug
Updating marvinroger/AsyncMqttClient          0.8.2 @ ^0.8.2                     [Up-to-date]

Library Storage: /home/user/dev/my_app/.pio/libdeps/my_app_release
Updating marvinroger/AsyncMqttClient          0.8.2 @ ^0.8.2                     [Up-to-date]

```

vslinuxdotnet, 2021-07-15:
+1

Hello @bastianhjaeger,
The problem is the new sdk of ESP that remove the openssl support (too big)

I Solve 99% of the problems:
In file platform.txt add: -lcrypto -laxtls in line compiler.c.elf.libs=
Get the libs: libcrypto.a and libaxtls.a and the ssl.h include from a 2.X SDK and put the libs in: packages/esp8266/hardware/esp8266/3.0.1/tools/sdk/lib and the ssl.h in packages/esp8266/hardware/esp8266/3.0.1/tools/sdk/include

The compile almost works, but.... the linking fail because the IRAM, don't have enough size available... :(
If you use a old ESP SDK 2.4.X it works, so the ssl must for ported to use the bearssl to work with the new ESP SDK 3.X.
That's it.

closed 2021-10-11 by stale[bot]

bastianhjaeger, 2022-01-10:
Unfortunately the issue subsists. Any solutions from anyone that works 100%?

bastianhjaeger, 2022-01-10:
@vslinuxdotnet Which `ssl.h` did you exactly use? Could you share a github link?

vslinuxdotnet, 2022-01-12:
> @vslinuxdotnet Which `ssl.h` did you exactly use? Could you share a github link?

Please read my post: https://github.com/me-no-dev/ESPAsyncTCP/issues/170#issuecomment-880507278

bastianhjaeger, 2022-01-12:
> > @vslinuxdotnet Which `ssl.h` did you exactly use? Could you share a github link?
> 
> Please read my post: [#170 (comment)](https://github.com/me-no-dev/ESPAsyncTCP/issues/170#issuecomment-880507278)

I did read your post. But it is unclear to me which repositoriy and which ssl.h file you took. I found multiple and they where not at al equal. 

It would be great if you could share a link to the github repo, branch and file.

murilopereirame, 2022-11-24:
The proposed solution will lead you to an Out of Space error.
The workaround is downgrade your Arduino Core to 2.6.3 at your platformio.ini file 

> platform = espressif8266@2.6.3

In 2.6.3 version, the ssl exists and doesn't eat all yout IRAM space.